jonnew/cyclops #3: Power switch installation instructions

jsiegle, 2016-04-01:
The power switch needs to be placed through the appropriate hole in the enclosure before being soldered to the PCB. The instructions should be updated to reflect this.

jonnew, 2016-04-01:
Are you looking at the [Kit assembly guide](https://github.com/jonnew/cyclops/blob/master/resources/beta-kit-getting-started-guide.md) or the [main manual](https://github.com/jonnew/cyclops/blob/master/MANUAL.md)? This is mentioned in the kit assembly manual (step 2 is the switch in the panel, step 4 is solder the switch to the board. Some of the initial kits were shipped without a slip of paper directing people to the beta kit manual and that might be the problem.

I would like to combine the kit assembly manual with the main manual soon.

jsiegle, 2016-04-01:
I was reading the main manual. Looks good in the kit assembly guide!

jonnew, 2016-04-01:
I also now see that the main manual has a picture of the switch soldered in before being put in the panel, which is what you were saying, and is wrong. I will fix that.

commit d0669759 (2016-04-04): Update MANUAL.*

- Fixes #3
- Partially merged the kit assembly instructions with the main manual
- Added a bunch of TODOs that generally concern interfacing the cyclops
  with external devices and LEDs, installing the Arudino, programming
  the Arrduino etc.
closed 2016-04-04 by jonnew