jonnew/cyclops #4: EMI shielding

jsiegle, 2016-04-01:
The manual recommends the use of conductive coating for EMI suppression. Is this really necessary?

jonnew, 2016-04-01:
Probably not in most situations. This is especially true if you are using a digital acquisition system, e.g. with Intan headstages and the cyclops is very far away from the analog portion of the circuit.

jsiegle, 2016-04-01:
I am, as it turns out

jonnew, 2016-04-01:
@jsiegle, sorry, I deleted your last comment when I clicked on it, can you repost?

jsiegle, 2016-04-01:
Just wanted to add that the power connector for the Arduino touches the bottom of the enclosure, so anyone using a conductive coating should be careful not to spray it there.

jonnew, 2016-04-01:
Ah, yes. I need to add this to the manual. When installing the arduino, I always remove the power jack using a hot air gun to prevent this.

jonnew, 2016-04-01:
Also to note: keeping cables short and low inductance is the best way to reduce EMI. I actually mount my LED right to the back of the device.

[image]
[image]