jonnew/cyclops #7: USB port on Cyclops betas

Callahanra, 2016-05-07:
Hi!
I have two cyclops kits and both are lacking the usb input port on the board. Do I need to install that myself? If so, I missed any reference to specs on what to use, do you have a recommendation?

Thanks!
Rebecca

jonnew, 2016-05-07:
Hi,

The cyclops is not supplied with an Arduino, which provides the USB input. To get access to that, you will need to get an Arduino Leonardo and install it on the cyclops during the assembly process.

Have you had a look at the getting started guide? Specifically, have a look at the "Kit Contents" section: https://github.com/jonnew/cyclops/blob/master/resources/beta-kit-getting-started-guide.md#kit-contents

jonnew, 2016-05-07:
Also, for general questions like this, feel free to email the open ephys forum: https://groups.google.com/forum/#!forum/open-ephys

That way the answer will be visible to a lot of other people that might also be confused. Let me know if this resolves your issues, BTW.

closed 2016-05-07 by jonnew

Callahanra, 2016-05-07:
Clears it up, thanks!