jonnew/cyclops #13: Overcurrent indicator always on

jsiegle, 2016-07-26:
I just assembled a Cyclops. The overcurrent indicator light turns on as soon as the module receives power and stays on indefinitely. I haven't tried hooking it up to an LED yet. Any idea what could be wrong?

jonnew, 2016-07-26:
Did you assemble it with the Arduino? If so, the LED is probably on because the default pin state corresponding to the overcurrent monitor is defaulted to HIGH. Try reprogramming the arduino with one of the [examples](https://github.com/jonnew/cyclops/tree/master/arduino/examples) and see if it turns off.

See this example for how to use the overcurrent protection feature:

https://github.com/jonnew/cyclops/blob/master/arduino/examples/over_current_protect/over_current_protect.ino

jsiegle, 2016-07-26:
Thanks, that fixed it.

closed 2016-07-26 by jsiegle