jonnew/cyclops #15: LED continuously on

jsiegle, 2016-08-12:
Whenever I turn on my Cyclops, there's a baseline LED current that won't go away. I can still modulate the signal with the Ext input, but the power never dips below the baseline level.

It was working fine last week, and I can't figure out what might have changed. Any suggestions for troubleshooting?

jonnew, 2016-08-12:
1. Are you referring to a measured baseline current? Does this translate to
perceiabel light being emitted from the LED? There is a baseline current
2. Physically remove the ext input. Does the problem go away? If so, your
ext input singal has a small non-zero voltage in its "off" position that
the cyclops is dutifully following
3. Move the source switch to the middle ("off") position. Still baseline
current?

jsiegle, 2016-08-12:
1. The emitted light is quite bright (about 50% of max). I'm not measuring the current output.
2. Same thing happens when Ext is unconnected.
3. Same thing happens regardless of the position of the source switch.
4. Grounding the Ext input doesn't fix it either

jonnew, 2016-08-12:
That seems like a real issue with your circuit in particular (at least I hope so). What does the voltage on VREF look like?

jsiegle, 2016-08-12:
Ref Out = -0.005 V
LED Curr = 0.083 V
Ext In = 0.59 V (with nothing connected)

Seems like the DC offset on Ext In is the issue. But what could be causing it?

jonnew, 2016-08-12:
OK interesting.

- Voltage VREF is a direct reflection, with unity gain, of the control signal going into the current driver's feedback loop (see final page of the schematic https://github.com/jonnew/cyclops/blob/master/cyclops/cyclops_schematic.pdf). When this is zero, the current should be zero.
- Voltage on Ext In may or may not reflect anything given the state of the source switch (e.g. it could be floating when that is set to OFF or DAC; pg.1 one of schematic).

My new guess is that the circuit may be oscillating. Plug LED Curr measurement into an oscope and see if this is the case.

jonnew, 2016-08-24:
Any updates on this?

jsiegle, 2016-09-14:
No, I just switched to a different Cyclops board, which has been working great. The broken one still has the same issue. I don't see any oscillations, just a DC offset on the LED current.

jonnew, 2016-09-14:
Steven Flavell just brought down with a similar issue and it seems like it might be a PCB defect, which is very troubling. The issue there was the the 1.25V rail is shorted onto GND. This eventually destroys the Cuk inverter by overheating it. When you loose your negative rail all bets are off. To see if you are having a similar issue:

When you have a minute:

- Pull the PCB out of the enclosure
- Apply power
- Test the -5V, -1.25V, 2.5V, 5V, and 12V test points. Report measured voltages.

jsiegle, 2016-09-15:
-5V = -5.047
-1.25V = -1.26
2.5V = 2.50
5V = 5.00
12V = 12.06

If I plug in an LED, -1.25V goes up to -1.2. Not sure if that matters.

jonnew, 2016-09-15:
OK, I don't understand this issue then. Feel free to ship the board to me if you're not using it and I will see if I can figure this out. If its something general, I really want to fix it before the next production round...

jsiegle, 2016-09-16:
Shipping it back to you, should arrive early next week.

jonnew, 2016-09-16:
Awesome, thanks for that!

jonnew, 2016-09-19:
OK, it seems the issue is that my voltage rail -1.25V to 12V exceeds the maximum rating for this chip by 0.25V. This is likely what I causing the failure. This is bad because it will affect all devices. I will likely email the list with this as a know issue. The solution is to remove the chip and tie the feedback signal directly to the current sense resistor, loosing the ability to do optical FB. I don't think many people are using that option anyway.

It is fixed in the latest revision by the addition of a 6V rail.

jsiegle, 2016-09-19:
Cool, glad you figured it out!

cdeister, 2017-08-07:
Hey Jon. I had the same issue crop up. Do you still recommend to tie the feedback signal to the current sense resistor? Or, perhaps, is there some other chip I could swap it out for? Thanks as always for your work on this kind of stuff!

jonnew, 2017-08-07:
Hi Chris. Sorry to hear that. Please have a look at this post to for a fix. Also, if it works for you, please let me know! 

https://groups.google.com/forum/#!searchin/open-ephys/cyclops%7Csort:relevance/open-ephys/cO1ZDqjeaAI/thvctO-iCwAJ

jonnew, 2017-10-21:
This issue is fixed in revision 3.6

closed 2017-10-21 by jonnew