jonnew/cyclops #22: Different functions when input pin is either RISING or FALLING

guidomeijer, 2020-09-30:
Hi Jon, I have a question. I want to have a pulse train going as long as the TRIG input is HIGH. I've tried doing this by using `set_trigger` to attach a function that starts the pulse train on the rising edge and another function that stops the pulse train on the falling edge. This didn't work out for me. I could only get one of the two functions to work, never both. And it doesn't seem like a very good way of doing this because I want to have the pulse train going as long as the trigger input is high (which varies from trail to trial). Is there a way to simply get whether the trigger is high or low in the loop() section?

guidomeijer, 2020-09-30:
Never mind, I figured it out, it was super easy. I just did `pinMode(TRIG0, INPUT); ` during `setup()` and then read it out with `digitalRead(TRIG0)`

closed 2020-09-30 by guidomeijer

jonnew, 2020-09-30:
yep that's a good solution. Just for the record, if you want to use the `set_trigger` function to do this, you can use the `mode` parameter `HIGH` as well. See the "parameters" section of the `attachInterrupt` docs, which is being used by this function

https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/?setlang=it

When in doubt, your solution is better because it does not deal with asynchronous operations which interrupts inevitably do.