jonnew/cyclops #24: LED warming up time course issue

polbech, 2021-05-25:
Hello! 

I am currently using two Cyclops to deliver alternated pulsed light (8ms pulses) from 2 LEDs for imaging. I have noticed in my images a consistent artifact in the normalized fluorescence where it seems that a considerable number of frames start from very negative values. When checking the raw fluorescent traces in non-biological samples, it seems that the intensity of one of the LEDs progressively ramps up over the initial ~ 20 frames. The other LED shows an opposing trend, starting up higher and ramping down over a longer time span. Initially I figured it might be from the controller but when I checked in the oscilloscope, the output pulses seem to work fine. Image corresponds to an example of the fluorescent traces in a non biological sample, x axis represents frames, and y axis is normalized fluorescence.

[image]

With my limited knowledge about LEDs, I have figured out that at least the ramping down trend could be a junction temperature issue, but I still don't know if the ramping up behaviour is expected, if it can be fixed by compensating with the driver pulses, or if I should just replace the LED altogether. Has anyone observed a similar type of behaviour with their LEDs? Can this be fixed, perhaps compensating by tweaking the output from the driver in the initial frames, or should I just replace the LED altogether? In case this is an intrinsic LED thing, what would be the best way to correct this to minimize the contamination during real acquisition?

Thank you very much in advance and let me know if I can provide any further information!
Best,
Pol

PS: I am using the Luxeon Star LEDs from Quadica attached to a cylindrical aluminum heat sink with fins spanning the whole perimeter.

jonnew, 2021-05-25:
Hi,

This is interesting. It definitely seems like a thermal issue, although, as you've noted, this typically is inversely related with the LEDs output. The long time constant seems to indicate that the effect is integrating due to the heat sink. 

A couple things:

- Do you have a a way to monitor the current output of the cyclops ("LED Curr") just to make sure you are actually getting correct constant current behavior over the course of the experiment? You can do this with an digital oscope by setting the trace persist to infinite and triggering on the rising edge of the current pulse. 
- If the device is operating normally (delivering constant current), then you will need temperature or optical compensation. You can get optical compensation (the device controls optical power output instead of current) by using the optical-feedback mode along with an amplified photodiode. See page. 14 of the manual.

jonnew, 2021-05-25:
Another easier option is to simply let the system reach thermal equilibrium before opening the shutter for the LEDs' light path.

polbech, 2021-05-27:
Hi,

Thank you very much for your reply. The current output of the cyclops seemed to be stable from the beginning as measured from the oscilloscope. As you suggested in your second comment, I figured a way around it by letting the LED run in a low power offset in the non recorded periods, so at least the increasing trend is gone and the more normal, decreasing trend is now present.

Thanks for the tips!