jonnew/cyclops #27: TRIG input remains at ~3V after receiving a single pulse

ThomasHainmueller, 2023-07-19:
Hi John, 

When we use an arduino digital output (or any other 5V trigger pulse for that matter) we run into the following issue: The cyclops is triggered once as expected (attached LED blinks) but then the voltage at the 'TRIG' BNC input remains at a level of 2.9-3V persistently. We were able to mitigate this somewhat by connecting the 'TRIG' bnc pin and shield with a 100 kOhm resistor to allow that voltage to dissipate but it's not ideal and really doesn't work at all when the trigger comes from a 3.3V arduino (e.g. due).

Do you have any idea where that residual voltage comes from and is there any way to zero it out in arduino software or otherwise?

Thanks a lot for your help!
Thomas

jonnew, 2023-07-19:
Sounds like a hardware issue. Where did you get this device?

jonnew, 2023-07-19:
Also trig is an input pin, how are you measuring it?

ThomasHainmueller, 2023-07-20:
Hi John,

Thanks a lot for the quick reply! The device has been in our lab for a bit. I'd have to ask around when/were it was bought, but it's not self-built. I have the same problem with two different cyclopses. I have a BNC with two separate leads attached and measure the voltage between the two with a voltmeter (see image below). The other ends of the cables go into the GND and Digital output pin of an arduino, respectively. The voltage is zero at baseline, rises to 5V when the pulse is given by the arduino but then stays at 2.9 and the cyclops can't be triggered anymore. With the resistor attached as shown on the photo, the voltage will settle back to zero after the pulse and the cyclops can be triggered repeatedly. Unfortunately I found no config where this would work with a 3.3V input pulse, mostly because that seems to leave only a very narrow range for resistors that would do the trick.

[image]

ThomasHainmueller, 2023-07-20:
The other ends are attached to the GND and Digital out pin of another Arduino (Due). However, I tried other 3.3V and 5V pulse generators instead and I get the same problem. The TRIG input on the cyclops will hold the 3V voltage even when the pulse generator is disconnected altogether, so it's unlikely to come from the input device. Also, when I attach the same input cable to the SIGNAL IN port on the cyclops, the voltage will go to zero when the input coming from the cables is set to zero.

jonnew, 2023-07-20:
Has any custom firmware been uploaded to the device that may have turned
those pins into outputs?

ThomasHainmueller, 2023-07-20:
To the best of my knowledge no (I 'inherited' the device). How would I re-upload the default firmware?

ThomasHainmueller, 2023-07-20:
Also, just to clarify: Does 'firmware' in this case mean the cyclops.h c++ library that's embedded in the teensy scripts or is there another piece of firmware that lives outside of that?

jonnew, 2023-07-20:
In the end, there is a teensy 3.2 in there that can be programmed in any
way (e.g. defining trig to be an output). You can use some of the example
scripts in there to get things back to normal if that has been changed.

https://github.com/jonnew/cyclops/tree/master/lib/cyclops/examples

ThomasHainmueller, 2023-07-20:
Unfortunately, that doesn't fix the issue. I just re-downloaded the entire library from github and re-uploaded the 'trigger pulse' example script, but I still get the same result. It doesn't make any sense to me, but it seems that somehow in this example the pin on the teensy stays high once it's triggered which prevents it from accepting further input until it's externally zeroed out...

jonnew, 2023-07-20:
OK. Sorry to hear that. At the end of the day, this is a copper trace connected to a pin on a micro controller. If that pin is programmed to be an input and is getting stuck at a voltage, then something seems like its wrong with the teensy. Perhaps an overvoltage condition of some sort occurred? These pins are 5V tolerant though.

You could get another Teensy 3.2 and replace the one thats in there if you want to see if that fixes your issue.

ThomasHainmueller, 2023-07-20:
No problem - I'm still somewhat optimistic we can figure this out. It's weird because the same problems comes up on three different cyclopses (just tried a third one) which are otherwise working fine. I was thinking to make some low-level tests with the built-in teensy to see if I can pin down the problem.

Is there a layout in the documentation that shows which pins of the teensy are connected to which in/outputs/actuators on the cyclops board and chassis? Maybe I'm blind but somehow I couldn't find it in the (otherwise excellent) manual. Thanks again for your help!

ThomasHainmueller, 2023-07-21:
...JK. Found it (cyclops_schematic.pdf).

ThomasHainmueller, 2023-07-21:
It's not a problem of your device directly, but others had issues with the pullup resistors of the teensy, too. 
https://forum.pjrc.com/threads/46709-Teensy-3-2-INPUT_PULLUP-Problems-with-Pushbutton
It also does not have built-in pull-down resistors, so it is likely necessary to attach one externally as shown in the image, otherwise the pin floats like it did for me. I checked and the problem is also present with a bare-bone code that only reads from the digital pin without using any of the cyclops libraries, so it is a hardware issue (of the teensy itself).

ThomasHainmueller, 2023-07-21:
...Oook - finally got it to work. The issue with the pull-down resistor is really only an issue if the other end of the cable isn't (functionally) connected to anything - e.g. with a mechanical switch or something of that sort. With the Arduino on the input side, the problem only arises if you fail to declare the triggering input from the other device as an output (I'm an idiot...) once you do, the problem is solved. Leaving this here just in case it's useful to anybody else.

ThomasHainmueller, 2023-07-21:
Also, thank you again Jon for your input and bearing with me while figuring this out.

jonnew, 2023-07-21:
Yeah as you are getting at, if you are driving the pin with a low output impedance source, like a function generator or the output pin on another microcontroller, then the pull down/up state of that pin should not matter.

Glad you figured it out, you're not an idiot, have a good day.